puhitaku/mtplvcap #11: Not working with D700

puhitaku, 2020-09-05:
From YasukiKishi:

 - D700を電源ON⇒mtplvcap.exe起動したらシャッターは開く。
 - localhost:42839でISO,F値,Focus制御はできるが、Width,Height,Frame rateの情報は取れていない、プレビューも絵が映らない
 - localhost:42839/viewも表示できない

```
[0000]  INFO usb: found: 04b0:0422
[0000]  INFO usb: found: 0cf3:e303
[0000]  WARN mtp: detected more than 1 device, opening the first device: 04b0:0422
[0000]  INFO main: started
[0002]  WARN lv: workerLV: failed to start live view: the camera is not ready
[0003]  WARN lv: frameCaptor: failed to decode header
```

puhitaku, 2020-09-05:
Related branch: [fix/d700](https://github.com/puhitaku/mtplvcap/tree/fix/d700)

puhitaku, 2020-09-08:
I got a clue: https://github.com/dukus/digiCamControl/blob/master/CameraControl.Devices/Nikon/NikonD700.cs#L42-L64
There exist some "Live View Modes"

Jokler, 2023-08-18:
Removing the magic 8 here makes video work for me: https://github.com/puhitaku/mtplvcap/blob/94f3b4c8849c347930e13e34d1c9a2c6fdb9864a/mtp/server.go#L741

puhitaku, 2023-08-19:
@Jokler Wow, great! I'll add a special path for D700 and release an experimental version later.

puhitaku, 2023-09-08:
@Jokler Sorry for keeping you waiting, here's an experimental build that includes the fix you proposed: [v1.6.2-rc1](https://github.com/puhitaku/mtplvcap/releases/tag/v1.6.2-rc1).
Could you try this build and see if it works? Thank you!

Jokler, 2023-09-08:
If I just try to run that I get this:
```
[0001] ERROR lv: failed to start live view (InvalidStatus). Investigating the reason...
[0001]  WARN lv: workerLV: failed to start live view, reason: recording destination is the card
```
I was testing on top of the [fix/d700](https://github.com/puhitaku/mtplvcap/tree/fix/d700) branch when it was working for me. If I run my working version first and then run [v1.6.2-rc1](https://github.com/puhitaku/mtplvcap/releases/tag/v1.6.2-rc1) keeping the camera connected it works too.

Since I wasn't super clear before only the video seems to work though, I'm not sure if the settings in the web interface are supposed to work for all cameras though.

puhitaku, 2023-09-09:
Some older models need us to change the "record media" to the RAM. The message looks like it.
I pushed a commit to `fix/d700-new` and created a corresponding release `v1.6.2-rc2`.
@Jokler Could you try this again and see if it works without invoking the old fix/d700 build? Thanks for your cooperation!

Extra info of `fix/d700`: the branch is really old, and had some experimental commits that are also implemented in the master branch in very similar but different lines of code. I just gave it away to rebase and solve conflicts due to too many conflicting diffs.

Jokler, 2023-09-10:
Now it works right away! :)

puhitaku, 2023-09-10:
I've just released [Version 1.6.2](https://github.com/puhitaku/mtplvcap/releases/tag/v1.6.2) :tada:
Now I can close here, the longest issue I've ever had. Thank you very much! @Jokler

closed 2023-09-10 by puhitaku