puhitaku/mtplvcap #12: d5300 don't work

viktor02, 2020-10-03:
Hi I am using your program with nikon d5300 however it prints errors and no image from the camera in the browser.
What could be the error?
Also, during startup, the program take a picture(judging by the sound), but after some time a timeout error appears and errors begin to appear
I also tried the nikon webcam utility, but I also have no camera image in it.

video: https://drive.google.com/file/d/1eRFpiXMSmE6y80bg5K282pCUKTHyPRKI/view?usp=sharing

puhitaku, 2020-11-14:
It seems like something is interfering with the communication between mtplvcap and the D5300.

Is there a utility related to backups like Dropbox, NAS backup tools, etc.? One of my users reported that Qsync (QNAP's backup utility) was interfering with mtplvcap on his/her computer.

puhitaku, 2020-11-14:
@viktor02 sorry for inactivity,

It seems like something is interfering with the communication between mtplvcap and the D5300.

Is there a utility related to backups like Dropbox, NAS backup tools, etc.? One of my users reported that Qsync (QNAP's backup utility) was interfering with mtplvcap on his/her computer.

viktor02, 2020-12-03:
> @viktor02 sorry for inactivity,
>
> It seems like something is interfering with the communication between mtplvcap and the D5300.
>
> Is there a utility related to backups like Dropbox, NAS backup tools, [...]

Now I tried on reinstalled Windows - the error is different:
[image]

It looks like my Chinese cable was broken :(

viktor02, 2020-12-03:
Also after reboot, error same as first post

puhitaku, 2020-12-07:
I guess it's an environmental issue that is identical to [this comment thread](https://www.reddit.com/r/Nikon/comments/hzcue6/software_to_turn_your_nikon_cam_into_a_webcam_on/gc90yo5/). I hope there's no software interfering with mtplvcap as you've re-installed Windows...

There's nothing more to suggest you to do but... how about changing the USB port to connect? Does other alternative software like Nikon Webcam Utility work?

puhitaku, 2021-01-25:
Going to close this issue. If you suffer additional issue, please open this or a new issue.

closed 2021-01-25 by puhitaku

viktor02, 2021-01-31:
Hi! I managed to find the original cable from nikon d750 and it worked! Nikon d750 has a resolution of 1024x768 and d5300 has 640x424. It's sad, but apparently these are Nikon's problems.

puhitaku, 2021-02-01:
@viktor02 That's great! Do you mean that you successfully used mtplvcap with D750? I'm facing issue #25 and the author says s/he could not use mtplvcap with D750. If you did, it's a big leap! I'd like to add D750 to "confirmed cameras" list with your name alongside.

viktor02, 2021-02-01:
Yes, it works, but as I understand it, it has Linux, and I ran it from Windows, maybe this is the problem

puhitaku, 2021-02-01:
Thanks for your cooperation, from the bottom of my heart

viktor02, 2021-02-02:
Sorry, I put it wrong when I talked about resolution. On nikon d750, I get the same error as here #25