puhitaku/mtplvcap #13: D7100 image is really dark ISO is 0

wifiuk, 2020-10-14:
image is really dark and when you go on the web-browser the ISO is set to 0.

When you change it, it basically crashes and nothing else works from that point on, have to kill the camera to quit the application.

puhitaku, 2020-11-14:
Some older camera may have a different way to exchange ISOs in MTP layer.

Adjusting the dial on your camera and choosing "Auto Mode" might solve this issue temporarily. We have no control of ISO in the Auto Mode and the camera adjusts it automatically.

puhitaku, 2020-11-14:
@wifiuk sorry for inactivity,
Some older camera may have a different way to exchange ISOs in MTP layer.

Adjusting the dial on your camera and choosing "Auto Mode" might solve this issue temporarily. We have no control of ISO in the Auto Mode and the camera adjusts it automatically.

puhitaku, 2020-12-07:
@wifiuk Ping

puhitaku, 2021-01-25:
I'll mark this issue outdated and going to close now. Please re-open if the problem persists.

closed 2021-01-25 by puhitaku