puhitaku/mtplvcap #16: D7000で以下メッセージが出て動作しませんでした。

akimitwo, 2020-10-22:
環境：macOSX Catalina 10.15.7
Nikon D7000 / TypeCアダプタ経由のUSB接続

git cloneでコンパイルして利用

@MBP15:~ $ ./mtplvcap -debug server
[0000]  INFO usb: found: 04b0:0428
[0000]  INFO usb: found: 0bda:8153
[0000]  WARN mtp: detected more than 1 device, opening the first device: 04b0:0428
[0000]  INFO main: started
[0000] DEBUG lv: manufacturer = NIKON, product = NIKON DSC D7000, serialnumber = 00000xxxxxxxxxx
[0000] DEBUG lv: model didn't match, falling back to the generic model%!(EXTRA string=)
[0001] DEBUG lv: current recording media: SDRAM
[0001] ERROR lv: failed to start live view (InvalidStatus). Investigating the reason...
[0001]  WARN lv: workerLV: failed to start live view, reason: the release mode is mirror-up

puhitaku, 2020-11-14:
Debug flag ついててありがたいです。"Invalid Status" で Live view が開始できない場合は、様々な理由でカメラ上の設定に問題があり開始できてないケースが多いです。

今回の場合、理由として "the release mode is mirror-up" と最後に出てます。下のリンクにあるように、D7000ではレリーズモードとしてミラーアップ撮影が可能です。これを通常の撮影モード（S）に戻してから再度お試しいただけますか？

https://onlinemanual.nikonimglib.com/d7500/ja/11_release_mode_03.html

puhitaku, 2020-11-14:
@akimitwo 気付いておらずすみません。
Debug flag ついててありがたいです。"Invalid Status" で Live view が開始できない場合は、様々な理由でカメラ上の設定に問題があり開始できてないケースが多いです。

今回の場合、理由として "the release mode is mirror-up" と最後に出てます。下のリンクにあるように、D7000ではレリーズモードとしてミラーアップ撮影が可能です。これを通常の撮影モード（S）に戻してから再度お試しいただけますか？

https://onlinemanual.nikonimglib.com/d7500/ja/11_release_mode_03.html

puhitaku, 2020-12-07:
@akimitwo こちらいかがでしょうか。

akimitwo, 2020-12-07:
せっかく返信いただいたのにスミマセン！！
CatalinaからBigSurへの環境移行で失念しておりました。

本日確認後、ご連絡します！

akimitwo, 2020-12-07:
@puhitaku
こんばんわ

Sモードにして確認してみたところ、前回とは異なる動きになりました。
コントロール画面（http://localhost:42839/）にはアクセスでき、
Periodic Auto Focusをenableにしたらフラッシュがあがりましたので操作は可能になっていました。
http://localhost:42839/viewのほうは、何も閲覧できませんでした。

ログではこのようになっていました。

11:35PM osmaniax@Z390Mac:~/mtplvcap $ ./mtplvcap -debug server
[0000]  INFO usb: found: 04b0:0428
[0000]  INFO usb: found: 0bda:8153
[0000]  INFO usb: found: 0a12:0001
[0000]  WARN mtp: detected more than 1 device, opening the first device: 04b0:0428
[0000]  INFO main: started
[0000] DEBUG lv: manufacturer = NIKON, product = NIKON DSC D7000, serialnumber = 000002077242
[0000] DEBUG lv: model matched: D700
[0001] DEBUG lv: current recording media: SDRAM
[0001]  WARN lv: frameCaptor: failed to decode header
[0002]  WARN lv: frameCaptor: failed to decode header
[0003]  WARN lv: frameCaptor: failed to decode header
[0004]  WARN lv: frameCaptor: failed to decode header
[0005]  WARN lv: frameCaptor: failed to decode header
[0006]  WARN lv: frameCaptor: failed to decode header
[0007]  WARN lv: frameCaptor: failed to decode header
[0008]  WARN lv: frameCaptor: failed to decode header
[0009]  WARN lv: frameCaptor: failed to decode header
[0010]  WARN lv: frameCaptor: failed to decode header
[0011]  WARN lv: frameCaptor: failed to decode header
[0012]  WARN lv: frameCaptor: failed to decode header

puhitaku, 2020-12-07:
これはバグっぽいですね。モデル検知にミスった結果、得られた画像の解析に失敗しているようです。
修正次第メンションいたします。少々（数日以内）お待ちいただけますでしょうか？

akimitwo, 2020-12-07:
なるほどです、ご確認ありがとうございます。
もちろんお時間のある時で大丈夫です！

puhitaku, 2020-12-10:
@akimitwo モデル検出の修正を入れた Version 1.1.3 をリリースしました。お試しいただけますか？ https://github.com/puhitaku/mtplvcap/releases/tag/v1.1.3

akimitwo, 2020-12-10:
確認しました！
バッチリ認識できており、View画面でも滑らかに動いています！

ご対応どうもありがとうございます！

closed 2020-12-10 by akimitwo