puhitaku/mtplvcap #18: Not working with D3000

okhiroyuki, 2020-12-02:
Mac Big Sur 11.0.1にて検証いたしました

## 状況

- D3000を電源ON => ./mtplvcap
- localhost:42839は表示されるが、各種設定に反応せず、プレビューも絵が映らない
- localhost:42839/viewも表示できない

## ログ

こちらでなにかわかりますでしょうか？

```
% ./mtplvcap
[0000]  INFO usb: found: 04b0:0424
[0000]  INFO main: started
[0000]  WARN lv: frameCaptor: failed to obtain an image: OperationNotSupported
[0001]  WARN lv: workerLV: failed to get live view status: DevicePropNotSupported
[0001]  WARN lv: frameCaptor: failed to obtain an image: OperationNotSupported
[0002]  WARN lv: workerLV: failed to get live view status: DevicePropNotSupported
[0002]  WARN lv: frameCaptor: failed to obtain an image: OperationNotSupported
[0002]  INFO http: GET / 127.0.0.1:51368
[0002]  INFO http: GET /assets/popper.min.js 127.0.0.1:51374
[0002]  INFO http: GET /assets/bootstrap4-toggle.min.css 127.0.0.1:51370
[0002]  INFO http: GET /assets/bootstrap4-toggle.min.js 127.0.0.1:51378
[0002]  INFO http: GET /assets/bootstrap.min.js 127.0.0.1:51376
[0002]  INFO http: GET /assets/jquery-3.5.1.slim.min.js 127.0.0.1:51372
[0002]  INFO http: GET /assets/bootstrap.min.css 127.0.0.1:51368
[0002]  INFO http: GET /favicon.ico 127.0.0.1:51368
[0003]  WARN lv: workerLV: failed to get live view status: DevicePropNotSupported
[0003]  WARN lv: frameCaptor: failed to obtain an image: OperationNotSupported
[0004]  WARN lv: workerLV: failed to get live view status: DevicePropNotSupported
[0004]  WARN lv: frameCaptor: failed to obtain an image: OperationNotSupported
[0004] ERROR lv: HandleControl: failed to set ISO: failed to set ISO: AccessDenied
[0004] ERROR lv: HandleControl: failed to set ISO: failed to set ISO: AccessDenied
[0005]  WARN lv: workerLV: failed to get live view status: DevicePropNotSupported
[0005] ERROR lv: HandleControl: failed to set ISO: failed to set ISO: AccessDenied
[0005] ERROR lv: HandleControl: failed to set ISO: failed to set ISO: AccessDenied
[0005]  WARN lv: frameCaptor: failed to obtain an image: OperationNotSupported
[0006]  WARN lv: workerLV: failed to get live view status: DevicePropNotSupported
[0006] ERROR lv: HandleControl: failed to set f-number: failed to set f-number: failed to stop live view: OperationNotSupported
[0006]  WARN lv: frameCaptor: failed to obtain an image: OperationNotSupported
[0006] ERROR lv: HandleControl: failed to set f-number: failed to set f-number: failed to stop live view: OperationNotSupported
[0007]  WARN lv: workerLV: failed to get live view status: DevicePropNotSupported
[0007] ERROR lv: HandleControl: failed to set f-number: failed to set f-number: failed to stop live view: OperationNotSupported
[0007]  WARN lv: frameCaptor: failed to obtain an image: OperationNotSupported
[0008]  WARN lv: workerLV: failed to get live view status: DevicePropNotSupported
[0008]  WARN lv: frameCaptor: failed to obtain an image: OperationNotSupported
^C[0009]  INFO signal: caught signal: interrupt
[0009] ERROR main: interrupt
```

okhiroyuki, 2020-12-05:
よく考えたら、このタイプはそもそも動画が取れないですね。

puhitaku, 2020-12-07:
@okhiroyuki D3000 はライブビュー撮影に対応していないので動作しないと思われます。`failed to get live view status: DevicePropNotSupported` とあることからライブビューに関する操作にも対応していないことが伺えます。

D3000 の動作情報提供ありがとうございました。

closed 2020-12-07 by puhitaku