puhitaku/mtplvcap #19: Nikon Z6 - No MTP Extensions in ''

IvanCharcos, 2020-12-22:
Not entirely sure what is going on.

Followed the Linux instructions and tried both pre-built and self-built approaches of the software.

Encountering the following error.

```
>> /mtplvcap -debug server
[0000]  INFO usb: found: 04b0:0443
[0000] ERROR mtp: fatal error LIBUSB_ERROR_IO; closing connection.
[0000] FATAL mtp: failed to detect MTP devices: could not open 04b0:0443: mtp: no MTP extensions in ''
```

```
>> lsusb
Bus 009 Device 006: ID 04b0:0443 Nikon Corp. 
```

Any help appreciated!

JoergSchi, 2021-01-10:
I use OSB-Studio (ubuntu 20.04.1) and install the pre-built mtplvcap for the Nikon Z6.
Play Attention to unmounted the camera before starting mtplvcap.

Following these Steps:
1. Nikon Z6 connected by USB-C to PC, camera get mounted
2. unmount the camera
3. start mtplvcap
4. ... (go on with the instructions)

I start mtplvcap from the terminal:

```
./mtplvcap
[0000]  INFO usb: found: 04b0:0443
[0000]  INFO main: started
```

puhitaku, 2021-01-25:
@IvanCharcos Sorry for inactivity,

Like JoergSchi suggests, if there are any applications and mounts that interfere mtplvcap, please unmount or remove them and try again.

closed 2021-04-19 by puhitaku