puhitaku/mtplvcap #22: NOT work on Win10 Pro(Oct./2020)

Higomon, 2021-01-21:
開発者様

お世話になります。

macOS Big Sur 11.1ではD800E/D7000ともに無事動いたので
味をしめてWin10機２台で試しました。

しかし、全く起動しなかったのでご報告いたします。

## 症状
### Win機 1台目(Dell PRECISION TOWER 7910)
前提: OSのゴミが溜まってるのか、少々状態が不安定

1. driverのinstallは無事に完了し、デバイスマネージャでも確認済み
2. Win用ビルド済みの実行ファイルを開いた瞬間に落ちる。

### Win機 2台目(Panasonic Let's Note CF-SZ5)
前提: つい最近、OSはクリーンインストール済み

1. driverのinstallは無事に完了し、デバイスマネージャでも確認済み
2. Win用ビルド済みの実行ファイルを開いたら、以下のlogまで出して落ちる
[0000] INFO usb: found: 04b0:042e
[0000] INFO usb: found: 8087:0a2b
[0000] WARN mtp: detected more than 1 device, opening the first device: 04b0:042e

## 試した環境
### OS
  Win10 Pro(Oct./2020) ver. 20H2
  OS build 19042.746
  
### Camera
- D800E 
- D7000
 
### App
- Win用ビルド済みの実行ファイル
- OBS 26.1.1-64bit Win
- Zoom 5.4.6(59296.1207)
- Firefox 84.0.2

puhitaku, 2021-01-25:
@Higomon mtplvcap のバージョンはどれをお使いかわかりますでしょうか？
この issue を作成された日付からすると 1.1.3 ではないかと思われますが。

Higomon, 2021-01-25:
mtplvcap v1.1.3を使いました。

最新のv1.2.0も試させて頂きました。
管理者としてmtplvcapを実行したところ、瞬時に落ちました。
試した２台のWin機で同じ症状でした。

puhitaku, 2021-02-07:
@Higomon 大変お待たせしました。瞬時に落ちるバグの原因がおそらく特定されました。以下お試しいただけますでしょうか。

README の[一眼レフのドライバを置き換える](https://github.com/puhitaku/mtplvcap/blob/master/README.ja.md#1-%E4%B8%80%E7%9C%BC%E3%83%AC%E3%83%95%E3%81%AE%E3%83%89%E3%83%A9%E3%82%A4%E3%83%90%E3%82%92%E7%BD%AE%E3%81%8D%E6%8F%9B%E3%81%88%E3%82%8B)のセクションで、Zadig 上で選択するドライバをもともと `libusb-win32 (vX.X.X.X)` としていましたが、これを `WinUSB (vX.X.X.X)` に変えて `Replace Driver` をクリックしてください。

インストールが完了したら、mtplvcap を起動してみてください。最新バージョンの v1.2.0 も動作するはずです。もしもっと新しいバージョンがリリースされていたら、その時点での最新をダウンロードして起動してください。

puhitaku, 2021-02-07:
[README.ja.md](https://github.com/puhitaku/mtplvcap/blob/master/README.ja.md#1-%E4%B8%80%E7%9C%BC%E3%83%AC%E3%83%95%E3%81%AE%E3%83%89%E3%83%A9%E3%82%A4%E3%83%90%E3%82%92%E7%BD%AE%E3%81%8D%E6%8F%9B%E3%81%88%E3%82%8B) をアップデートし、WinUSB をインストールするように書き換えました

Higomon, 2021-02-08:
ご対応いただき、ありがとうございます。
インストールするドライバやOSの影響が大きかったようです。
以下、ご報告いたします。

---
### ご報告
1) 上記Win機２台にて検証し、ご指示の通りにWinUSBドライバに変更したところ、
Win機１台(※)ではmtplvcap v1.1.3およびv1.2.0とのに正常動作を確認しました。
　※検証前に、OSはクリーンインストール済み

2) もう一台のWin機では動作いたしませんでした。
    (mtplvcap v1.1.3 / v1.2.0ともに)

3) 1.で使用したWin機にて、カメラとの接続確立後、
mtplvcapを終了しても、カメラ側が通信しっぱなしの状態を継続しており、
mtplvcapを再度起動しても接続に失敗します。
カメラを再起動すれば通信に成功します。

puhitaku, 2021-02-10:
一台では動いたようでなによりです。もう一台はどのようなエラーが出ますか？また `-debug mtp,server` というオプションを追加するとどのように出力されますか？よくある不具合としては [USB 通信に干渉するソフトがある](https://github.com/puhitaku/mtplvcap/issues/12#issuecomment-727216169)というのがあり、確認いただきたいです。

Higomon, 2021-02-10:
カメラと通信できないWin機で以下のように試しました。

オプションの付け方があってるか、わかりかねますが、
いかがでしょうか？ご検証をお願いします。

### 所感
- mtplvcap.exe(※)を右クリック→「管理者として実行」すると即落ち
　※v1.1.3/v1.2.0ともに同じ症状

- コマンドプロンプト(管理者モード)でmtplvcap.exeを実行すると、
以下のようなログを出力

### driver replacement
[image]

### command prompt | mtplvcap_v1.1.3
[image]

### command prompt | mtplvcap_v1.2.0
[image]

puhitaku, 2021-02-10:
オプションの付け方は合ってますね。
カメラかもしれないデバイスとして認識されたものが複数あり、そのうちの最初のもの（`8087:07dc`）を使っているのが問題のようです。以下のように実行して、Nikon のカメラと思われる `04b0:042e` を指定してみてください。

`mtplvcap.exe -vendor-id 0x04b0`

この `04b0` を優先する実装はまだ入っておらず、今後追加予定です。

Higomon, 2021-02-10:
ご指示の通りのコマンドで無事に動作を確認できました。
誠にありがとうございます。
感動です！
これでZoom等でキレイな画を写せて、嬉しい限りです。

---
### 認識したカメラ以外のデバイス
- 8087:07dc: 後付で拡張したIntel製WiFi/Bluetooth
[image]

- 0b95:1790: 有線LAN×2 port？
[image]

puhitaku, 2021-02-10:
良かったです！では問題なく動作したということで、この Issue は Close させていただきます。報告ありがとうございました！

closed 2021-02-10 by puhitaku