puhitaku/mtplvcap #25: D750 support

skontrolle, 2021-01-27:
Hi, I have tried with a D750 on linux but did not succeed in opening the shutter.

```
[0000]  INFO usb: found: 04b0:0437
[0000]  INFO main: started
[0000] DEBUG lv: manufacturer = NIKON, product = NIKON DSC D750, serialnumber = XXXXX
[0000] DEBUG lv: model didn't match, falling back to the generic model%!(EXTRA string=)
[0001] DEBUG lv: current recording media: card
[0001] DEBUG lv: the recording media is the card. Switching it to the SDRAM.
[0004] ERROR mtp: fatal error LIBUSB_ERROR_TIMEOUT; closing connection.
[0007]  WARN lv: frameCaptor: failed to obtain an image: mtp: cannot run operation MTP_WMPPD_PlaylistObjectPref, device is not open
```

The `-backend-go` gets rid of the USB error timeout but does not open the shutter either.  Let me know if you would like some more info or to try something else.

puhitaku, 2021-01-30:
@skontrolle  Thanks for reporting,

It looks like some communications were successful while it finally failed in an earlier stage.

Could you check if your D750 shows up [PC] in the LCD like in [this comment](https://github.com/puhitaku/mtplvcap/issues/6#issuecomment-669324754)? It'd ensure where it stopped working.

Memorandum (please ignore):

> falling back to the generic model

It looks okay since the fallback model and D750 have same header size and communication should success

skontrolle, 2021-01-31:
> Could you check if your D750 shows up [PC] in the LCD like in this comment? It'd ensure where it stopped working.

With `-backend-go` it does say [PC] instead of the number of files but using the standard libusb does not.

skontrolle, 2021-01-31:
Actually, it does briefly go to [PC] mode with libusb after the 0001 but resets when it hits the timeout.

```
[0001] DEBUG lv: the recording media is the card. Switching it to the SDRAM.
[0004] ERROR mtp: fatal error LIBUSB_ERROR_TIMEOUT; closing connection.
[0007]  WARN lv: workerLV: failed to start live view: LIBUSB_ERROR_TIMEOUT
```

puhitaku, 2021-02-07:
After some days I still have no idea about this behavior because the timeout occurs with no hints and signs, and only raises a generic LIBUSB_ERROR_TIMEOUT. If some conditions don't meet the camera should raise InvalidStatus. A timing problem can be a cause but not confident. I keep this issue opened until I get a chance to debug it with a real D750.

Note: according to #12 it might be a D750 specific problem

skontrolle, 2021-02-07:
#12 is at least a good sanity check.  Are there any more verbose debugging options I can try to help, maybe in libusb?

I also tried to get it to run on an older buster laptop but the version of go was too old to compile and use some of the functions in mtplvcap.

puhitaku, 2021-04-29:
@skontrolle Hi, I wish you notice this mention. I got a chance to borrow a D750 and finally found the cause of this issue. It was a quirk or a bug of D750 software that it stops responding after commencing specific command. I'll publish the fixed mtplvcap sooner.

closed 2021-04-29 by puhitaku

puhitaku, 2021-04-29:
Released the new version: https://github.com/puhitaku/mtplvcap/releases/tag/v1.2.1

skontrolle, 2021-05-08:
Thanks and verified that it worked!

I suspect that bug might be more prolific as I'm able to get it to stop working by trying to change the aperture from the web interface.

puhitaku, 2021-05-13:
The camera won't work as you expect when the mode is other than "M" (manual) which enables you to control all parameters like aperture etc.

PR #33 (2021-04-29): Mitigate D750 quirk
Fix #25