puhitaku/mtplvcap #26: dyld: Library not loaded: /usr/local/opt/libusb/lib/libusb-1.0.0.dylib on Apple Silicon computer

javag97, 2021-02-22:
Also tried to build myself using source:
```
 CGO_CFLAGS='-Wno-deprecated-declarations' go build .
```
Where I get: Package libusb-1.0 was not found in the pkg-config search path.

I see that brew installled version `1.0.24` on my computer, so I believe that's the problem. How would I go about fixing it?

puhitaku, 2021-02-27:
@javag97 I successfully reproduced the issue on my MacBook Pro (Late 2020, M1). Since building it manually succeeds on my Mac, I didn't notice that the distributed binary does not work. Sorry for the inconvenience. I added a hand-crafted ARM64 build in the latest release.

Have you tried `brew link`? This will put symlinks where pkg-config can find.

```
$ brew link libusb
```

puhitaku, 2021-03-11:
@javag97 Any update? I hope you got it working 😉

javag97, 2021-03-23:
Sorry for the delay! Let me try :-)

closed 2021-04-09 by puhitaku