puhitaku/mtplvcap #28: Feature: Arch Linux AUR package

Artiom-M, 2021-03-24:
Hi!
Thank you for this wonderful tool! Worked like a charm with D7000.
I've created an AUR package for Arch Linux 
https://aur.archlinux.org/packages/mtplvcap-git/

puhitaku, 2021-04-01:
@Artiom-M  Oh my god, I'm so proud to see it in AUR! Thank you for using mtplvcap and adding it to AUR!

Since I'm not sure about the AUR nature, let me know how to upgrade the AUR'ed mtplvcap when I pushed the new version. Thanks in advance.

Artiom-M, 2021-04-06:
>  [data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAADAAAAAwCAYAAABXAvmHAAAABHNCSVQICAgIfAhkiAAABCBJREFUaIHtmT1yGzcUx3+g6LTRpEwjqnGb9QlE1ckm5AmoHMAjyuMD+ACJJfoCkk5ASmvXXJ4gy9KVKLWZ8TitteJLsSDNBRa74IfkzIT/Gc1oAbwv4AH44xG2+LZQj6VYIiRnKHwcW7XHUPqU2AbwrbGxvJQ+DZ5xwJQARQA0jSExQkKNhHtGqs1kE3bXDkDe00FoAa0lRQcoBuoXLtexv3IAEtEEToFgHQeABDhRIfEqwisFINecouiuIutWypn6lZNlxZYKQPrsUqePnd8zjFFc6Fz/rH4mAZAPBEzZ1XIt4CeHfExKW7X57OuTdwDa+SHFKXNJyhvfjSl9GtR5A3QKuhNSDn2D8A8gYog987fA0ar5q/fRBbBndMUq5NBHh9c9IBFn2M6PSAlWdR5AhcSkBMDI6Gpqm9U6qgboWRoazSMVOvfBSpCIGDgwmg+rJshnBU6N71vSpc/8amQ6bytsWyhdAbnmCMW50Vw6K9KnwQ7HKJp83fAJQswDvbKN7ljttgoZuGTKV6BmzfRlqfMRZ9S50XfE4mkVoOhS50au3bOqdedvZsVRmYvOFdBH3U2uMWXfNYMSkeA+300kKuTFJuy6V2DH2qTjEufP8HceIHCdMtrGuMKXOdwBKOPCUlwUDZM+DeDYaL5D8Zrsouro/++MMcdatgj5nDd9WUDZHsgLSUYLLNQtTnSH4oQpHxE+IXxiykcUJ5hB2LIzxKW+5FTM/DPesBZqzqu9mftSvGPKF2vUlC/UeIfwx0Kree5/tTXNtRy43tjeL7IZMSuAmfs3haMAhL+NlsKZLbFl4WmflML9plV6ByAfnHk4Nr73S6z9WCFbZcvCfA+YdRuLm2R8vggxi2kkvKTGK2sf1PgO4WWBrA3blpN7la1APg9dR1lqned7CG9RPEfxg/57jvAWkzbbsjOYzjr3RN3VgZDk1kTogG1QtZlIRI/8XbAH/OnUnaGMF/1m+eKAewUe7LPYdfGokC6OfHZgrGUsaBv51bZ9mcMZgGozQXGVa8yegcXjQwKg5+pfQE+PLUbdIHuKqzIGW34KTS0a29GUtxAqpEvKPlkgiysyBnqk7LtmHuZ0Os+ApZjCzG2WdWqleZapmHDPi2UqBz6QPrs84y8kl6aVLz+feyA/Y0JDl1Y2izpDw3nAnbIzeFUlNPU1GefSNZxC3e5yTa8s3WZYpqwSY5IvxQTh97XKKorzgpn3Lhr4c6Hs0Z0/KjPDQ3nPeQm3tyB9GhLRh8K0GS9TNFiltDjARYOzG3OAELPDP7nS4gPf64d+Cze/H5HSepTS4iIce2JdeOW8iXXL68u+hYswBrpPWl5fhES0UBwhBn+ptnyFcFFW8/FTsyHoglZTs9YAe5+MyApcCQ/E/5mfmFzY/k7siW0AW/zf8S/Eo2PIp3W8vwAAAABJRU5ErkJggg==###24x24:true###][Tracking image][https://github.com/notifications/beacon/AA665PYHFYWNSOC2SN3YFKDTGR5LVA5CNFSM4ZWOQVAKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOGBSSTHA.gif]

AUR is source based packaging system. The package I've created points to your master branch. So each time a user builds the package it takes the latest master.
It can have some drawbacks if your master branch is not stable by your branching strategy. In this case we can consider other options.

puhitaku, 2021-04-09:
> It can have some drawbacks if your master branch is not stable

master branch is considered to be stable so it's not a problem. Thank you for the details!

closed 2021-04-09 by puhitaku