puhitaku/mtplvcap #30: workerLV: failed to start live view: the camera is not ready

denied404, 2021-04-05:
I am trying to get `mtplvcap` working with my Nikon D5000 camera. It is connected to my desktop via USB, `0x0423` its ID (I have more than one MTP devices connected, therefore I specify that explicitly). Whenever and whereever I tried to run the software (I have `Windows 10` and `Arch Linux` in dual-boot) I have one and the same error:
```
➜ mtplvcap -product-id 0x0423
[0000]  INFO usb: found: 04b0:0423
[0000]  INFO main: started
[0002]  WARN lv: workerLV: failed to start live view: the camera is not ready
```
I do not have any possible interfering software running or any other drivers for my `D5000` on both OS and still I continue to getting one and the same error.
I see that `D5000` is in the list of supported cameras, just wandering what am I missing...

puhitaku, 2021-04-12:
@denied404  The cause of "camera is not ready" error is unclear and it's not reproducible. Please let me know if the battery is sufficient, if an SD card is inserted or not, the USB cable is an official one or not, and if it works after resetting the DSLR.

puhitaku, 2021-04-12:
[I pushed the experimental fix](https://github.com/puhitaku/mtplvcap/tree/fix/notready) of this problem. While I'm not sure if it works, could you build this branch and run it with debug output enabled? `mtplvcap -product-id 0x0423 -debug mtp,server`

denied404, 2021-04-12:
@puhitaku thanks for the answer! I am getting the error with SD Card inserted, 100% battery and official USB cable.
The funny thing is: when I got the error for the very first time I have immediately reported about the issue here and hadn't even tried to check camera stream. And then I occasionally did and it seems that despite the fact that I am still having the error - everything works fine (except the fact that LiveView actually does not start on the camera, but I can live with that).
But I will build your experimental branch and run with debug output and then report back to you! Just for the sake of getting rid of all errors :)

puhitaku, 2021-04-19:
ping @denied404

denied404, 2021-04-19:
@puhitaku I have tried `fix/notready` branch (sorry for the delay!). What I see is:
```
➜ ./mtplvcap -debug server -product-id 0423           
[0000]  INFO usb: found: 04b0:0423
[0000]  INFO main: started
[0000] DEBUG lv: manufacturer = NIKON, product = NIKON DSC D5000, serialnumber = 000006445579
[0000] DEBUG lv: model matched: D5000
[0001] DEBUG lv: current recording media: SDRAM
[0002] DEBUG lv: current recording media: SDRAM
[0004] DEBUG lv: current recording media: SDRAM
[0005] DEBUG lv: current recording media: SDRAM
[0007] DEBUG lv: current recording media: SDRAM
[0008] DEBUG lv: current recording media: SDRAM
[0010] DEBUG lv: current recording media: SDRAM
```
And the camera constantly opens and closes shutter. Live view by http://localhost:42839/view works, but the picture constantly freezes and those freezes are aligned with open/close shutter events