puhitaku/mtplvcap #31: Not working with D300

mochihisa, 2021-04-11:
I tried to use it on my D300, but it did not work.
```
[0000]  INFO usb: found: 04b0:041a
[0000]  INFO main: started  
[0002]  WARN lv: workerLV: failed to start live view: the camera is not ready  
[0003]  WARN lv: frameCaptor: failed to decode header  
```
- I heard the shutter open.
- localhost:42839/view was not displayed.
- After changing the F-stop on localhost:42839 and disconnecting the connection, I checked the settings of the main unit and found that the value had been changed.

The environment in which it was run is as follows.

OS : Ubuntu 20.10
Version : [v1.2.0](https://github.com/puhitaku/mtplvcap/releases/tag/v1.2.0)

puhitaku, 2021-04-12:
Hi,

The shutter won't open if "the camera is not ready" error happens except you change the F value. I'd like to ask you to confirm that it opened after you changed it manually.

On the other hand, [I pushed the experimental fix](https://github.com/puhitaku/mtplvcap/tree/fix/notready) of this problem. Can you build this branch and run it with debug output enabled? `mtplvcap -debug mtp,server`

puhitaku, 2021-04-19:
ping @mochihisa

mochihisa, 2021-04-19:
I'm sorry for the late reply. I don't have time right now, so I'll reply later.

mochihisa, 2021-04-25:
I'm getting different output than before. Is there any possibility?

```
[0000]  INFO usb: found: 04b0:041a
[0000] ERROR mtp: fatal error LIBUSB_ERROR_IO; closing connection.
[0000] FATAL mtp: failed to detect MTP devices: could not open 04b0:041a: mtp: no MTP extensions in ''
```

puhitaku, 2021-05-11:
@mochihisa Sorry for inactivity,
It looks like the camera is not responding in an earlier stage than the original error message (failed to decode header). `no MTP extensions in ''` indicates many types of fundamental problems like interfered USB communication, broken USB cables, etc. Please checkout the master branch and see if the problem persists or not.

mochihisa, 2021-05-16:
I'm sorry, I'll be late in replying due to periodic exams.

mochihisa, 2021-06-06:
Sorry for the late reply.
The same result was obtained using the master branch. It seems that the cable is not damaged because it can be used for photo transfer.

yanorei32, 2022-04-12:
I had a same problem with D300.

Environment:
Nikon D300 firmware A: 1.01, B: 1.00
libusb 1.0.25-3
Arch Linux 5.17.1-arch1-1

Log (Release 1.5.0 and 3bc52d45 are showed same logs):
```
$ ./mtplvcap -debug mtp,server
[0000]  INFO usb: found: 05ac:821f
[0000]  INFO usb: found: 04b0:041a
[0000]  WARN mtp: detected more than 1 device
[0000]  INFO mtp: opening the detected Nikon DSLR: 04b0:041a
[0000]  INFO main: started
[0000] DEBUG lv: manufacturer = NIKON, product = NIKON DSC D300, serialnumber = 000002014808
[0000] DEBUG lv: model matched: D300
[0001] DEBUG lv: current recording media: SDRAM
[0002]  WARN lv: workerLV: failed to start live view: the camera is not ready
[0003]  WARN lv: frameCaptor: failed to decode header
[0004]  WARN lv: frameCaptor: failed to decode header
[0005]  WARN lv: frameCaptor: failed to decode header
```

And I tried to change f-number but I got ERROR.
```
$ ./mtplvcap -debug mtp,server
...
[0010] DEBUG lv: HandleControl: set ISO: 2500
[0011]  WARN lv: frameCaptor: failed to decode header
[0015]  WARN lv: frameCaptor: failed to decode header
[0016] DEBUG lv: HandleControl: set f-number: 10
[0016] ERROR lv: HandleControl: failed to set f-number: failed to set f-number: RetCode a005
[0016] DEBUG lv: current recording media: SDRAM
[0017]  WARN lv: workerLV: failed to start live view: the camera is not ready
[0018]  WARN lv: frameCaptor: failed to decode header
```