puhitaku/mtplvcap #32: D7000 wrong resolution reported

Artiom-M, 2021-04-19:
For D7000 the resolution reported is 4,928 x 3,264 (picture resolution) while actual live view is 640 × 424.
Is this the only possible resolution?
Camera is capable of  1,920 × 1,080 (24p) in video.

puhitaku, 2021-04-19:
Some older models have quirks that report the original image resolution. The dimension is parsed not from the actual JPEG frame but from Nikon's model-specific header.
Live view has no function to change the resolution, unfortunately.

closed 2021-04-19 by puhitaku

puhitaku, 2021-04-19:
While I closed this issue because this problem is a minor and non-critical one, this quirk should be mitigated in some ways like to decide the dimensions by parsing the actual JPEG frames

puhitaku, 2021-04-19:
I think I'll add a feature like exporting a raw (JPEG + MTP headers) and ask @Artiom-M to upload it. Could you cooperate with me? 😉

Artiom-M, 2021-04-19:
Sure, I will be happy to help