puhitaku/mtplvcap #35: Not Working with D7100

srimaln91, 2021-05-09:
Hi,

I'm using a go binary built in my PC on Ubuntu 20.04. But getting the following error.

`failed to detect MTP devices: could not open 8087:0a2b: LIBUSB_ERROR_ACCESS`

Kindly let me know if there is a way to resolve this.

puhitaku, 2021-05-11:
@srimaln91 There is Nikon DSLR `04b0:0430` but it's not selected. [Please run it with `-vendor-id` option](https://github.com/puhitaku/mtplvcap#usage) and see if it works.

```sh
$ ./mtplvcap -vendor-id 04b0
```

BTW I've pushed the new v1.3.0 that has an auto-selection feature that mitigates this issue.

Please reopen this if your problem persists.

srimaln91, 2021-05-12:
Thanks @puhitaku. The issue is resolved when I use vendor-id parameter.

Thanks a lot. 👍👍