puhitaku/mtplvcap #38: fatal error LIBUSB_ERROR_TIMEOUT; closing connection

tadasv, 2021-05-16:
Hey, I'm trying to get this working on debian with latest build from master but I'm getting this error:

```
ERROR mtp: fatal error LIBUSB_ERROR_TIMEOUT; closing connection
```

I have D5600, have you seen similar issues? What would be a good way to approach the issue?

I tried extending ModelMap by copying D5500 to D5600 config; still nothing.

puhitaku, 2021-05-16:
Do something happen when you run mtplvcap? e.g. shutter open

Please run it with `-debug server` and upload the entire output.

```
$ mtplvcap -debug server
```

puhitaku, 2021-05-16:
FYI: the parameters and implementation to communicate with D5600 seem to be [the same](https://github.com/dukus/digiCamControl/blob/b9b920bf8ea55d4ea1b38533f2b052b01a31357b/CameraControl.Devices/CameraDeviceManager.cs#L164) with D5200 ~ D5500.

When there's no corresponding Modelmap item, mtplvcap will fall back to the modern setting that is equal to D5200 ~ D5500.

tadasv, 2021-05-16:
@puhitaku no, the shutter does not open. I turn on camera, put it in live view. When I plug in the usb cable it just turns off the LV. Then I can't get the LV back on until I unplug the cable. Below is the output when running the tool. It seems to me that this is some USB connection issue, though I get the same issue when running with `sudo`. Btw, this is a patched version where I just copied D5500 to D5600, but I get the same with a fallback.

```
[0000]  INFO usb: found: 04b0:043f
[0000]  INFO usb: found: 8087:0aaa
[0000]  WARN mtp: detected more than 1 device
[0000]  INFO mtp: opening the detected Nikon DSLR: 04b0:043f
[0000]  INFO main: started
[0000] DEBUG lv: manufacturer = NIKON, product = NIKON DSC D5600, serialnumber = 0000003051502
[0000] DEBUG lv: model matched: D5600
[0001] DEBUG lv: current recording media: card
[0001] DEBUG lv: the recording media is the card. Switching it to the SDRAM.
[0004] ERROR mtp: fatal error LIBUSB_ERROR_TIMEOUT; closing connection.
[0007]  WARN lv: workerLV: failed to start live view: LIBUSB_ERROR_TIMEOUT
[0007]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
[0007]  WARN lv: frameCaptor: failed to obtain an image: mtp: cannot run operation MTP_WMPPD_PlaylistObjectPref, device is not open
[0008]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
[0008]  WARN lv: frameCaptor: failed to obtain an image: mtp: cannot run operation MTP_WMPPD_PlaylistObjectPref, device is not open
[0009]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
[0009]  WARN lv: frameCaptor: failed to obtain an image: mtp: cannot run operation MTP_WMPPD_PlaylistObjectPref, device is not open
[0010]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
[0010]  WARN lv: frameCaptor: failed to obtain an image: mtp: cannot run operation MTP_WMPPD_PlaylistObjectPref, device is not open
[0011]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
[0011]  WARN lv: frameCaptor: failed to obtain an image: mtp: cannot run operation MTP_WMPPD_PlaylistObjectPref, device is not open
[0012]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
[0012]  WARN lv: frameCaptor: failed to obtain an image: mtp: cannot run operation MTP_WMPPD_PlaylistObjectPref, device is not open
^C[0012]  INFO signal: caught signal: interrupt
[0013] ERROR main: interrupt
```

puhitaku, 2021-05-16:
I found that it's similar to a D750 bug that I fixed recently. It has a "quirk" that interrupts the USB connection after mtplvcap commences "recording media switch". I'm not sure if it's a quirk or a deprecation of API tho.

So I've pushed [a branch named `experiment/d5600`](https://github.com/puhitaku/mtplvcap/tree/experiment/d5600). Please pull the origin, checkout to it, and see if it works.

tadasv, 2021-05-16:
@puhitaku nice! I tried this out and it worked. It opens the shutter, captures video can see it via web interface.

closed 2021-05-16 by tadasv

puhitaku, 2021-05-16:
Good! Now is the time to close here while I have to get proceed: release the new version.

@tadasv And if it's okay with you, let me mention you in the README with the (glorious!) confirmation of D5600.

tadasv, 2021-05-16:
@puhitaku mention sounds good. I might bring up more issues if I encounter any as I continue with the camera setup.

puhitaku, 2021-05-23:
Released [v1.3.1](https://github.com/puhitaku/mtplvcap/releases/tag/v1.3.1)! Thank you for your cooperation again and have good video streaming with mtplvcap!