puhitaku/mtplvcap #40: fatal error LIBUSB_ERROR_TIMEOUT; closing connection

wadetregaskis-linkedin, 2021-05-26:
D5500, that works fine with Nikon's own software (Webcam Utility) other than the obnoxious black borders all around the frame, which is why I wanted to try this.

brew install libusb went fine.

Running the binary (directly from Terminal) does not prompt for permission in any way, and nothing shows up in the Security & Privacy system prefs pane.

All I get output is:

```
[0000]  INFO usb: found: 04b0:0438
[0000]  INFO main: started
[0004] ERROR mtp: fatal error LIBUSB_ERROR_TIMEOUT; closing connection.
[0007]  WARN lv: frameCaptor: failed to obtain an image: mtp: cannot run operation MTP_WMPPD_PlaylistObjectPref, device is not open
[0007]  WARN lv: workerLV: failed to start live view: LIBUSB_ERROR_TIMEOUT
[0007]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
[0008]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
[0008]  WARN lv: frameCaptor: failed to obtain an image: mtp: cannot run operation MTP_WMPPD_PlaylistObjectPref, device is not open
[0009]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
```

…and that repeats endlessly.

macOS 11.4 (20F71).  ("Big Sur")

puhitaku, 2021-05-31:
@wadetregaskis-linkedin
Does the shutter open? Please share the debug output of `./mtplvcap -debug server` .

fedegratti, 2021-06-02:
Hello @puhitaku, same is happening to me with a D3500, this is the output for `./mtplvcap -debug server`
[image]

wadetregaskis-linkedin, 2021-06-02:
No issues with a D500, only the D5500.  The shutter does not open on the D5500 - there's no indication at all, from the D5500 itself, that anything's happening.

```
👾 ./mtplvcap -debug server                                                                                                                                                                      [ 2:41:27 pm Wed  2/6/21]
[0000]  INFO usb: found: 04b0:043c
[0000]  INFO main: started
[0000] DEBUG lv: manufacturer = NIKON, product = NIKON DSC D500, serialnumber = xyz
[0000] DEBUG lv: model didn't match, falling back to the generic model%!(EXTRA string=)
[0001] DEBUG lv: current recording media: card
[0001] DEBUG lv: the recording media is the card. Switching it to the SDRAM.
^C[0024]  INFO signal: caught signal: interrupt
[0024] ERROR main: interrupt

👾 ./mtplvcap -debug server                                                                                                                                                                 [ 2:41:53 pm Wed  2/6/21]
[0000]  INFO usb: found: 04b0:0438
[0000]  INFO main: started
[0000] DEBUG lv: manufacturer = NIKON, product = NIKON DSC D5500, serialnumber = xyz
[0000] DEBUG lv: model matched: D5500
[0001] DEBUG lv: current recording media: card
[0001] DEBUG lv: the recording media is the card. Switching it to the SDRAM.
[0004] ERROR mtp: fatal error LIBUSB_ERROR_TIMEOUT; closing connection.
[0007]  WARN lv: workerLV: failed to start live view: LIBUSB_ERROR_TIMEOUT
[0007]  WARN lv: frameCaptor: failed to obtain an image: mtp: cannot run operation MTP_WMPPD_PlaylistObjectPref, device is not open
[0007]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
[0008]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
[0008]  WARN lv: frameCaptor: failed to obtain an image: mtp: cannot run operation MTP_WMPPD_PlaylistObjectPref, device is not open
[0009]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
[0009]  WARN lv: frameCaptor: failed to obtain an image: mtp: cannot run operation MTP_WMPPD_PlaylistObjectPref, device is not open
[0010]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
```

puhitaku, 2021-06-03:
Ugh, it seems to be a known issue like #38 .
This affects every recent DSLR!? Oh my...

Anyway, please wait for a while and I'll push an experimental build 🙏

puhitaku, 2021-06-16:
@wadetregaskis-linkedin @fedegratti 
Hi guys sorry for being late,
I've published an experimental build [Version 1.4.0 RC2](https://github.com/puhitaku/mtplvcap/releases/tag/v1.4.0-rc2). The source code is in branch [fix/media](https://github.com/puhitaku/mtplvcap/tree/fix/media). Could you try this one?

fedegratti, 2021-06-16:
Hi @puhitaku, thank you for your work. I tried this but I get the following error:
`dyld: Library not loaded: /usr/local/opt/libusb/lib/libusb-1.0.0.dylib`
I tried to install libusb in different ways and the v1.3.1 arm works (with the error we are discussing in this issue).

Maybe it is because I'm on an arm mac and the 1.4.0 RC2 has only an x64 binary?
Thank you

puhitaku, 2021-06-17:
Sorry for the inconvenience, I added an arm64 build.
https://github.com/puhitaku/mtplvcap/releases/tag/v1.4.0-rc2

fedegratti, 2021-06-17:
@puhitaku it worked!
[image]

puhitaku, 2021-06-17:
Yay! Thanks for checking.

This build is confirmed at least for D3500 -- and hopefully for D5500.
I'll release this as the stable build in advance of @wadetregaskis-linkedin 's reply and leave here opened.

puhitaku, 2021-06-17:
D3500 is a newly confirmed DSLR so I'll list it in README with your name! @fedegratti

fedegratti, 2021-06-17:
@puhitaku one thing to notice is that the connection is broken if I try to change the ISO and the aperture from `http://localhost:42839/`.
Autofocus and Rate limit are working fine!

fedegratti, 2021-06-17:
Here's the `./mtplvcap -debug server` output:
[image]

puhitaku, 2021-06-17:
Changing ISO and aperture is somewhat unstable also on my D5300 and some incompatible API changes may be introduced in recent models.

@fedegratti Is the operating mode M (manual) mode? If it is, the problem is definitely on mtplvcap's side.

fedegratti, 2021-06-17:
@puhitaku yes, it is in M mode, by the way, the ISO and aperture are actually performed in the camera! I mean, for instance, I have 1600 ISO, then I change it to 800 from the website, then the connection is broken but if I see the ISO in the camera, it is 800. Same for the aperture. So the API changes seem to work, but for some reason, the connection gets lost.

puhitaku, 2021-06-25:
The new version is released and closing here.

closed 2021-06-25 by puhitaku