puhitaku/mtplvcap #42: Z5 successfully working

ag-eitilt, 2021-06-24:
My Nikon Z5 successfully connects using mtplvcap-1.3.1 on Gentoo (5.10.27 kernel).  The OBS packaging doesn't support building the browser source component until the still-officially-unreleased 27.0.1 EBUILD, and even then I can only get Firefox to recognize the (built-in, not plugin) virtual camera and not the Chromium-based Brave.  Trying to manually set the F-stop throws `ERROR lv: HandleControl: failed to set f-number: failed to set f-number: AccessDenied`, and I don't know if I tested the ISO slider, but the other controls all work.

puhitaku, 2021-06-25:
Thank you for reporting 😉 I should inspect the F-stop manipulation sometime.
I'll update the README.

puhitaku, 2021-06-25:
I've updated the README. Thanks again, and enjoy using mtplvcap 😋

closed 2021-06-25 by puhitaku