puhitaku/mtplvcap #44: How to get other qualities?

Revisto, 2021-07-03:
Hi, I'm using Nikon D5100 and everything works fine but there is a problem. mtplvcap only captures 640x426 video quality. I wanted to see how I can get other qualities like full-HD as well.

puhitaku, 2021-07-03:
Dup: #34
Unfortunately, there's no other choice of resolution AFAIK.

kendermag, 2021-11-26:
Hello, i might found a way for this in another repo, here:

[https://github.com/dognotdog/ptpwebcam/blob/fd96bc3aaf5b5b02a43dd5d71e100b2ed95c1e17/PtpWebcamDalPlugin/PtpWebcamPtp.h#L122](https://github.com/dognotdog/ptpwebcam/blob/fd96bc3aaf5b5b02a43dd5d71e100b2ed95c1e17/PtpWebcamDalPlugin/PtpWebcamPtp.h#L122)

in this line there is a hex which is unused in your code in const.go , i assume writing the appropriate value there could be used in some cameras to increase the live view resolution, hence increasing the usb transferred resolution. The maximum resolution seems to be 1024x768 but that is way better than 640x424. 
Here is the commit where the other repo's owner added this feature: [commit](https://github.com/dognotdog/ptpwebcam/commit/aeac7f6406703c53fa07187455e5c588c7cac53f)

I tried to port this yesterday, but i am unskilled both in go and obj-c

puhitaku, 2021-11-26:
Interesting, opening it again

puhitaku, 2021-11-27:
I've tested it (read the available resolutions) and my D5300 returned 1 (320x212) and 2 (640,424).
Unfortunately, my DSLR isn't capable of another choice "3" (1024x?) but some may be.

puhitaku, 2021-11-27:
D5300 returns an array `[1, 2]` as available choices and I believe that newer models return like `[1, 2, 3]`. I'll try reading the choice from other models and post the update here.

puhitaku, 2021-11-28:
My expectation was right; on-production models return `[1, 2, 3]`. I tried it in a camera shop. Unexpectedly, I met a member of Nikon and he told me that MTP commands and properties are slightly different between models. Actually, some models like Z* reply `[1, 2, 3]` for the read value of property `0xD1AC` but writing it didn't succeed.

Changing resolution may fail but we can fallback so I'll implement like "try to read available resolutions and set it maximum and just log an error if it's not successful and go ahead".

kendermag, 2021-11-28:
Thank you! On the other repo it is stated that d7500 works with XGA here: [issue](https://github.com/dognotdog/ptpwebcam/issues/37)
What is interesting regarding a Z50 is here: [issue z50](https://github.com/dognotdog/ptpwebcam/issues/22) the person there says that he used Z50, and he got 720p resolution.

puhitaku, 2021-12-13:
I've created a PR: #52 
I wonder if I should enable this option as standard, but decided not to because the resolution change API is different for some models and I don't know how many models it would affect.

puhitaku, 2021-12-13:
I've released [Version 1.5.0](https://github.com/puhitaku/mtplvcap/releases/tag/v1.5.0) that has `-max-resolution` option. While there seems to be another unknown way to set it max on some models, I implemented it anyway and I should close here. Thanks @kendermag for pinging me with nice information!

closed 2021-12-13 by puhitaku