puhitaku/mtplvcap #45: Feature request: Send stream to stdout

Artiom-M, 2021-08-13:
I am using your program with OBS as per instruction as a webcam, but using OBS for just this is an overkill.
Would it be possible to send the same stream you are sending to websocket to the STDOUT? In this case, on Linux I would be able to pipe it to a virtual video device directly.

puhitaku, 2021-08-13:
While it makes sense to me, I currently have no idea what kind of container to pack the video stream for having it transmitted on stdout. It should be clarified, and it'll be more feasible when I have muxers written in Go.

Artiom-M, 2021-08-14:
You can just stream the same raw data as you streaming through the web socket then pipe it through ffmpeg, for example. So you don't really require a muxer. I would propose to make a switch --stdout which will just send captured stream to stdout instead of starting webserver.

maxammann, 2022-01-22:
I did a POC here: https://github.com/puhitaku/mtplvcap/commit/806a48fb0d5ff63114d4eaccfcf9900165452f98

With this command you can stream to basically anything:

`
CGO_CFLAGS='-Wno-deprecated-declarations' go run . -max-resolution -debug usb,data,mtp,server | ffmpeg -i - -vcodec rawvideo -pix_fmt yuv420p -threads 0 -f sdl Preview
`