puhitaku/mtplvcap #46: process doesn't stop when camera is disconnected

Artiom-M, 2021-08-14:
When camera is not connected and I start mtplvcap it is exiting, but if I just disconnect the camera while capturing it continue and just sending errors continuously:
```
[0054]  WARN lv: workerLV: failed to get live view status: LIBUSB_ERROR_NO_DEVICE
[0054]  WARN lv: frameCaptor: failed to obtain an image: mtp: cannot run operation MTP_WMPPD_PlaylistObjectPref, device is not open
[0055]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
[0055]  WARN lv: frameCaptor: failed to obtain an image: mtp: cannot run operation MTP_WMPPD_PlaylistObjectPref, device is not open
[0056]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
[0056]  WARN lv: frameCaptor: failed to obtain an image: mtp: cannot run operation MTP_WMPPD_PlaylistObjectPref, device is not open
[0057]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
[0057]  WARN lv: frameCaptor: failed to obtain an image: mtp: cannot run operation MTP_WMPPD_PlaylistObjectPref, device is not open
[0058]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
[0058]  WARN lv: frameCaptor: failed to obtain an image: mtp: cannot run operation MTP_WMPPD_PlaylistObjectPref, device is not open
[0059]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
[0059]  WARN lv: frameCaptor: failed to obtain an image: mtp: cannot run operation MTP_WMPPD_PlaylistObjectPref, device is not open
[0060]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
[0060]  WARN lv: frameCaptor: failed to obtain an image: mtp: cannot run operation MTP_WMPPD_PlaylistObjectPref, device is not open
[0061]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
[0061]  WARN lv: frameCaptor: failed to obtain an image: mtp: cannot run operation MTP_WMPPD_PlaylistObjectPref, device is not open
[0062]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
[0062]  WARN lv: frameCaptor: failed to obtain an image: mtp: cannot run operation MTP_WMPPD_PlaylistObjectPref, device is not open
[0063]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
[0063]  WARN lv: frameCaptor: failed to obtain an image: mtp: cannot run operation MTP_WMPPD_PlaylistObjectPref, device is not open
[0064]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
[0064]  WARN lv: frameCaptor: failed to obtain an image: mtp: cannot run operation MTP_WMPPD_PlaylistObjectPref, device is not open
[0065]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
[0065]  WARN lv: frameCaptor: failed to obtain an image: mtp: cannot run operation MTP_WMPPD_PlaylistObjectPref, device is not open
[0066]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
[0066]  WARN lv: frameCaptor: failed to obtain an image: mtp: cannot run operation MTP_WMPPD_PlaylistObjectPref, device is not open
[0067]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
[0067]  WARN lv: frameCaptor: failed to obtain an image: mtp: cannot run operation MTP_WMPPD_PlaylistObjectPref, device is not open
[0068]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
[0068]  WARN lv: frameCaptor: failed to obtain an image: mtp: cannot run operation MTP_WMPPD_PlaylistObjectPref, device is not open
[0069]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
[0069]  WARN lv: frameCaptor: failed to obtain an image: mtp: cannot run operation MTP_WMPPD_PlaylistObjectPref, device is not open
```

puhitaku, 2021-08-17:
I'm planning a huge improvement including this issue after I find the time. I'm now super busy so it'll be in October or November. Thank you.

Artiom-M, 2021-08-20:
I am using it in systemd unit that is started automatically by udev when camera is connected. I can share them if you are interested, but without this enhancement it will work only first time, then you need to kill it manually...

roguh, 2025-09-15:
> main function has a loop that reopens the connection, preserving any parameters 

I tried this code and encountered null pointer errors in `mtp/server.go` after reassigning a new USB device to `LVServer.dev`.

It seems more of the server's code needs to handle reopening logic, or perhaps the reopening logic can belong to `mtp/device_direct.go` so that the existing code can seamlessly keep using `LVServer.dev` after a reconnection.

puhitaku, 2025-09-16:
I must admit that the code will need a huge change to reconnect flawlessly. The changes should be divided into multiple PRs that improve the backend, the LV server, and the Web frontend. Maybe removing the "direct" backend, which relies on an unmaintained binding, will make the code slimmer and easier.

roguh, 2025-09-16:
> The changes should be divided into multiple PRs that improve the backend, the LV server, and the Web frontend.

I created an issue for an improved frontend: #94. This can be a standalone change.

> Maybe removing the "direct" backend, which relies on an unmaintained binding, will make the code slimmer and easier.

I think it's ok to have both. Perhaps switching to gousb by default is better, but maybe some systems work better with the old libusb.