puhitaku/mtplvcap #48: ERROR mtp: fatal error LIBUSB_ERROR_TIMEOUT on D810

Squitor, 2021-09-25:
Hi, i have encountered a similar issue on my D810 as some Users before me. 

```
./mtplvcap -debug server
[0000]  INFO usb: found: 04b0:0436
[0000]  INFO main: started
[0000] DEBUG lv: manufacturer = NIKON, product = NIKON DSC D810, serialnumber = 0000002009398
[0000] DEBUG lv: model didn't match, falling back to the generic model 
[0003] ERROR mtp: fatal error LIBUSB_ERROR_TIMEOUT; closing connection.
[0006]  WARN lv: frameCaptor: failed to obtain an image: LIBUSB_ERROR_TIMEOUT
[0006]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
[0006]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
[0007]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
[0007]  WARN lv: frameCaptor: failed to obtain an image: mtp: cannot run operation MTP_WMPPD_PlaylistObjectPref, device is not open

```
The shutter stays closed and the mirror up mode doesn't do anything either.

puhitaku, 2021-10-01:
@Squitor Thanks for reporting an issue. Which version did you use?

Squitor, 2021-10-01:
I used version 1.4.1 and also tried both fixes. All came with a similar result.

puhitaku, 2024-03-13:
Resolved in #80

closed 2024-03-13 by puhitaku