puhitaku/mtplvcap #53: Find the way to enable the higher resolution on all models

puhitaku, 2021-12-13:
Prop `0xD1AC` is the only way I know to change the resolution. Some models support reading/writing this. Some models only support reading. And others do not support neither reading nor writing.

According to a Nikon guy, there are many differences in MTP commands and props between DSLRs. Their [public SDK](https://sdk.nikonimaging.com/apply/) may give me something interesting, but it's licensed under a proprietary agreement that is considered not to be compatible with GPL.

I'm having conversation with them and I may implement this feature after they released it licensed under a OSI-approved licenses.

kendermag, 2021-12-14:
Something popped into my mind but i am not sure it is really a concern. So far the heating up was not an issue even for extended periods of time. Is it possible that with the resolution increase it might cause problems?

puhitaku, 2021-12-17:
Any change might cause problems (generally speaking)
According to the guide of D5300, it will shut the shutter when it gets too hot while LVing. This should apply to other models.

puhitaku, 2022-11-10:
D3S: replies `[1, 2]` but setting 2 stops the LV working