puhitaku/mtplvcap #55: SIGSEV with binaries and own build

nblumoe, 2022-01-16:
Thanks for providing the tool!

I tried the 1.5.0 binary and a local build from latest master. With both I get a SIGSEGV error when running the tool after granting it permissions via Security & Privacy. I'm using macOS 11.5.2 with an i7.

```
$ ./mtplvcap -debug server

fatal error: unexpected signal during runtime execution
[signal SIGSEGV: segmentation violation code=0x1 addr=0xb01dfacedebac1e pc=0x7fff20338c9e]

runtime stack:
runtime: unexpected return pc for runtime.sigpanic called from 0x7fff20338c9e
stack: frame={sp:0x7ffeefbff508, fp:0x7ffeefbff558} stack=[0x7ffeefb805b8,0x7ffeefbff610)
```

puhitaku, 2022-01-17:
Thanks for reporting the issue.
It seems like it's happening in the Go runtime.
Memo: the use of CGO can lead to this kind of crash (I can't avoid using this since the USB communication depends on libusb)

@nblumoe
Let me know:

 - What version of Go did you use?
 - Does the crash change if you plug or unplug a Nikon?

puhitaku, 2022-02-22:
@nblumoe Update?

nblumoe, 2022-02-22:
Hey @puhitaku, sorry I didn't have the chance to test it again and I'm not sure if I will be. Feel free to close the issue if you cannot reproduce it or think it isn't relevant.

puhitaku, 2022-02-22:
Okay, now I can't reproduce this since the cause is unclear but someone might suffer this problem again so I'm closing here temporarily. Reopen here if you and anybody like to.

closed 2022-02-22 by puhitaku