puhitaku/mtplvcap #56: "-max-resolution" not workig with Nikon Z6

maxammann, 2022-01-22:
Related: https://github.com/puhitaku/mtplvcap/issues/44

Switching to XGA does not work with mtplvcap. I also tested this with ptpwebcam on a Mac which actually worked.

I uploaded the camera info from ptpwebcam as `camera.txt`.

This is from the log of mtplvcap
[camera.txt](https://github.com/puhitaku/mtplvcap/files/7918620/camera.txt)

```
[0001]  INFO lv: available resolutions (higher is larger): [1 2 3]
[0001]  INFO lv: automatically use the largest choice: 3
[0004]  WARN lv: failed to change the image resolution (%!s(MISSING)); if it affects capturing frames, consider disabling `-max-resolution
```

maxammann, 2022-01-22:
In fact I found out what the problem is :)

On the Z6, the datatype is u8:

```go
	payload := struct {
		Resolution uint8
	}{
		Resolution: uint8(3),
	}
```

maxammann, 2022-01-22:
POC: https://github.com/puhitaku/mtplvcap/commit/2f455e1608eb2fad955bfd760312053ab34e205d

puhitaku, 2022-01-24:
Many thanks for the report and POC 👍 
I'll merge your changes somehow, like adding a quirk flag ;)
I leave here opened until then.

maxammann, 2022-01-24:
Are you sure this is a quirk? With which Nikons did you test the 64bit value?

maxammann, 2022-01-24:
> D500, D850, and D7500 want 64bit value and I confirmed it works.
> If the newer models incl. Z6 and the brothers want 8bit, it's not a quirk specific to Z6 and I'll add a new value like `ResolutionValueType` in the `mtp.ModelMap` struct.
> 
> (Extra story: I confirmed the max resolution feature in a camera store with permission from a clerk and I found that it doesn't work with Z6 and similar models. I didn't have a courage to ask clerk for a permission to debug my code in the store so I left there 😅)

Haha, nice! Do you know that you can get the documentation from nikon for all cameras? :) Im not sure whether one is allowed to use them though.

puhitaku, 2022-01-25:
> Do you know that you can get the documentation from nikon

That's what I'm negotiating with Nikon for. [Here](https://github.com/puhitaku/mtplvcap/issues/53) is the story. 

> Their public SDK may give me something interesting, but it's licensed under a proprietary agreement that is considered not to be compatible with GPL.

puhitaku, 2022-02-21:
Sorry for being inactive,
The fix is merged in #57 . mtplvcap now transmits an uint8 value for the newer models incl. Z6.

maxammann, 2022-02-21:
No worries :) thank you for your project!

Will test next week

fedegratti, 2022-02-24:
Hi @puhitaku I can confirm that this doesn't work on D3500:
[image]

maybe #57 fixes it?

Thank you for this project, I love it!

puhitaku, 2022-02-24:
Thanks for the report fedegratti-san,
I suspect that all Nikon needs uint8 (the fix introduced in #57) in the end so I'm going to test it in a week or so.
Please wait for a while and I'll mention you.

Ria8651, 2022-03-12:
I get exactly the same log as fedegratti with a D3400.

When using master I get:
[image]

Awesome project thanks :)

puhitaku, 2022-03-25:
Thank you everyone, now I identified the bug in changing the resolution and it is due to my untested change (Sorry for that! I have no new DSLRs capable of higher resolutions), so I fixed it now in #59 after a brief test in near camera shop.

Interestingly, the internal model name is different from what I expected e.g. Z7 II returns `NIKON DSC Z 7_2` as its model name. And D3400 also differs too (its model name is `NIKON DSC D3400` according to it while I expected "D3400").

fedegratti, 2022-03-25:
Hi @puhitaku, thank you so much for your support!

I've just tested and it still not working on D3500:
[image]

I hope the above image helps. Also, if you need me to do some testing on my camera, we can do chat about it. 

Lastly, do you have a place for donations? I love your work and I would like to give you something!

Cheers.

puhitaku, 2022-03-25:
That's because the resolution value type is "falling back" (as you see in the log) to uint64 I guess and I added a tweak to use uint8 by default. 

@fedegratti Could you pull the branch `fix/resolution` and give it a try again?

> do you have a place for donations? I love your work and I would like to give you something!

Thank you very much! Donation is very appreciated. The top page of mtplvcap and my GitHub profile page have a link to GitHub Sponsors and Patreon.

fedegratti, 2022-03-25:
> @fedegratti Could you pull the branch fix/resolution and give it a try again?

@puhitaku same issue. Is there a way to log more info? Maybe I can add more logs in the core.

puhitaku, 2022-03-28:
Argh, I temporarily can't think of what to do next ... I should borrow a D3500 (or other model that causes the "closing connection").

maxammann, 2022-03-28:
> Argh, I temporarily can't think of what to do next ... I should borrow a D3500 (or other model that causes the "closing connection").

We could also take a look at the specification. Im pretty sure thats legal.

maxammann, 2022-03-29:
According to some other project the D3500 is not capable of higher resolutions: https://github.com/dognotdog/ptpwebcam/blob/9f138716eaf1d14a3505f57b71480ab8577150e5/CAMERAS.md#aps-c--dx

keeandamok, 2022-04-20:
for the d3300 im getting a similar error to the d3400
[image]

puhitaku, 2022-06-18:
@maxammann 
> According to some other project the D3500 is not capable of higher resolutions

If its max is 640x480, it actually is. But they described the resolution pair with parens. I'm not sure what it means.

@fedegratti Unfortunately, no :P 

> Is there a way to log more info?

mtplvcap is not so verbose even when all logging options are enabled because I preferred to debug it with an interactive debugger. I think I should improve and increase the log...
At present, I'm pretty not sure how to remedy this and looking for a D3500 body.

puhitaku, 2022-06-18:
Sorry for neglecting here. And thanks for kind donations.

puhitaku, 2022-07-26:
@maxammann It's been a while, Z6 is confirmed to accept high resolutions according to a comment in #59 . The fixed version [v1.5.1](https://github.com/puhitaku/mtplvcap/releases) is released. Thanks for providing me a POC!

closed 2022-07-26 by maxammann

JobLeonard, 2022-09-20:
Despite the issue being closed I assume this is the best topic to give resolution feedback for untested models, is that correct?

I just tried this out for my `Z fc`. It works, max resolution is 1024x680. So a higher resolution is available, but it doesn't seem to have the full range the camera should be able to support.

[image]

JobLeonard, 2022-09-20:
Ah, so they distinguish the live view from their own "webcam" support? That explains it :)

Luckily 1024x680 is more than enough for my purposes, the image is so clean it's probably better than most people's HD webcams anyway ;)

And as a Linux user, thanks for writing this amazing piece of software!

evanslify, 2024-10-14:
Hello,

I confirm Nikon Z 8 is working with high resolution, with `-max-resolution` enabled.

```
$ ./mtplvcap -debug server -max-resolution
[0000]  INFO main: started
[0000] DEBUG lv: manufacturer = NIKON, product = NIKON DSC Z 8, serialnumber = ****
[0000] DEBUG lv: model didn't match, falling back to the generic model 
[0001]  INFO lv: getting available resolutions
[0001]  INFO lv: available resolutions (higher is larger): [1 2 3]
[0001]  INFO lv: automatically use the largest choice: 3
```

puhitaku, 2024-10-17:
@evanslify Thank you for reporting it!

gronktonkbabonk, 2026-01-17:
> I get exactly the same log as fedegratti with a D3400.
> 
> When using master I get: [image]
> 
> Awesome project thanks :)

did you ever get it working?

This is with the same model, d3400
[image]

puhitaku, 2026-01-17:
Nikons prone to timeout when the operation (changing resolution) is not supported. I find it weird too, but I have no clue at this point since I have no D3400.

gronktonkbabonk, 2026-01-17:
> > > I get exactly the same log as fedegratti with a D3400.
> > > When using master I get: [image]
> > > Awesome project thanks :)
> > 
> > 
> > did you ever get it working?
> > This is with the same model, d3400 <img alt="Image" width="996" height="445" src="https://private-user-images.githubusercontent.com/104763225/537068448-14aac88b-4835-411d-8229-a88d32c1a9e5.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Njg2NTY4OTEsIm5iZiI6MTc2ODY1NjU5MSwicGF0aCI6Ii8xMDQ3NjMyMjUvNTM3MDY4NDQ4LTE0YWFjODhiLTQ4MzUtNDExZC04MjI5LWE4OGQzMmMxYTllNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjYwMTE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI2MDExN1QxMzI5NTFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hMjZmZTViOWZjYWQ3OTg5ZGIwZDQzZDk3MjA5OTA2ZWI1YWQ5NjQ1OWI1MTc5Y2I2YTY0ZWVjOTgzZThhNGFlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.K_2xYnw7tsLkCEHWMnU6HGuFgW_W9IjNdL-SyJnJQu8">
> 
> Nope. I just tried building `master` branch with latest changes and I get the same output as you with my D3500. Also, I tried doing some changing and adding logs but without any luck. It's weird that says there are 2 available resolutions but is not possible to choose any.

> Nikons prone to timeout when the operation (changing resolution) is not supported. I find it weird too, but I have no clue at this point since I have no D3400.

Ah. Well i appreciate it anyway, still a great project :)