puhitaku/mtplvcap #60: Doesn't build

Ricotan-Linus, 2022-06-20:
I try build from source, but output many error.
but can't finish.
How to fix?
Sorry write poor English...

```
ricotyan:mtplvcap ricotan_aikatsu$ go version
go version go1.18.3 darwin/amd64

ricotyan:mtplvcap ricotan_aikatsu$ CGO_CFLAGS='-Wno-deprecated-declarations' go build .
go: downloading github.com/google/gousb v2.1.0+incompatible
go: downloading golang.org/x/sync v0.0.0-20200625203802-6e8e738ad208
go: downloading github.com/sirupsen/logrus v1.6.0
go: downloading github.com/x-cray/logrus-prefixed-formatter v0.5.2
go: downloading github.com/gorilla/websocket v1.4.2
go: downloading github.com/hanwen/usb v0.0.0-20141217151552-69aee4530ac7
go: downloading github.com/paulbellamy/ratecounter v0.2.0
go: downloading go.uber.org/atomic v1.6.0
go: downloading github.com/mgutz/ansi v0.0.0-20200706080929-d51e80ef957d
go: downloading golang.org/x/crypto v0.0.0-20190308221718-c2843e01d9a2
go: downloading golang.org/x/sys v0.0.0-20200519105757-fe76b779f299
go: downloading github.com/mattn/go-colorable v0.1.7
go: downloading github.com/mattn/go-isatty v0.0.12
# golang.org/x/sys/unix
../go/pkg/mod/golang.org/x/sys@v0.0.0-20200519105757-fe76b779f299/unix/syscall_darwin.1_13.go:29:3: //go:linkname must refer to declared function or variable
../go/pkg/mod/golang.org/x/sys@v0.0.0-20200519105757-fe76b779f299/unix/zsyscall_darwin_amd64.1_13.go:27:3: //go:linkname must refer to declared function or variable
../go/pkg/mod/golang.org/x/sys@v0.0.0-20200519105757-fe76b779f299/unix/zsyscall_darwin_amd64.1_13.go:40:3: //go:linkname must refer to declared function or variable
../go/pkg/mod/golang.org/x/sys@v0.0.0-20200519105757-fe76b779f299/unix/zsyscall_darwin_amd64.go:28:3: //go:linkname must refer to declared function or variable
../go/pkg/mod/golang.org/x/sys@v0.0.0-20200519105757-fe76b779f299/unix/zsyscall_darwin_amd64.go:43:3: //go:linkname must refer to declared function or variable
../go/pkg/mod/golang.org/x/sys@v0.0.0-20200519105757-fe76b779f299/unix/zsyscall_darwin_amd64.go:59:3: //go:linkname must refer to declared function or variable
../go/pkg/mod/golang.org/x/sys@v0.0.0-20200519105757-fe76b779f299/unix/zsyscall_darwin_amd64.go:75:3: //go:linkname must refer to declared function or variable
../go/pkg/mod/golang.org/x/sys@v0.0.0-20200519105757-fe76b779f299/unix/zsyscall_darwin_amd64.go:90:3: //go:linkname must refer to declared function or variable
../go/pkg/mod/golang.org/x/sys@v0.0.0-20200519105757-fe76b779f299/unix/zsyscall_darwin_amd64.go:105:3: //go:linkname must refer to declared function or variable
../go/pkg/mod/golang.org/x/sys@v0.0.0-20200519105757-fe76b779f299/unix/zsyscall_darwin_amd64.go:121:3: //go:linkname must refer to declared function or variable
../go/pkg/mod/golang.org/x/sys@v0.0.0-20200519105757-fe76b779f299/unix/zsyscall_darwin_amd64.go:121:3: too many errors
# pkg-config --cflags  -- libusb-1.0
Package libusb-1.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `libusb-1.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libusb-1.0' found
pkg-config: exit status 1
# pkg-config --cflags  -- libusb-1.0
Package libusb-1.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `libusb-1.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libusb-1.0' found
pkg-config: exit status 1
ricotyan:mtplvcap ricotan_aikatsu$ brew install libusb
Warning: You are using macOS 10.13.
We (and Apple) do not provide support for this old version.
You will encounter build failures with some formulae.
Please create pull requests instead of asking for help on Homebrew's GitHub,
Twitter or any other official channels. You are responsible for resolving
any issues you experience while you are running this
old version.

==> Downloading https://github.com/libusb/libusb/releases/download/v1.0.26/libusb-1.0.26.tar.bz2
==> Downloading from https://objects.githubusercontent.com/github-production-release-asset-2e65be/15120676/ebba8859-25d1-4a37-b80f-cf18699d2bdd?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220620%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Da
######################################################################## 100.0%
==> ./configure --prefix=/usr/local/Cellar/libusb/1.0.26
==> make install
🍺  /usr/local/Cellar/libusb/1.0.26: 22 files, 465.3KB, built in 14 seconds
==> Running `brew cleanup libusb`...
Disable this behaviour by setting HOMEBREW_NO_INSTALL_CLEANUP.
Hide these hints with HOMEBREW_NO_ENV_HINTS (see `man brew`).
ricotyan:mtplvcap ricotan_aikatsu$ CGO_CFLAGS='-Wno-deprecated-declarations' go build .
# golang.org/x/sys/unix
../go/pkg/mod/golang.org/x/sys@v0.0.0-20200519105757-fe76b779f299/unix/syscall_darwin.1_13.go:29:3: //go:linkname must refer to declared function or variable
../go/pkg/mod/golang.org/x/sys@v0.0.0-20200519105757-fe76b779f299/unix/zsyscall_darwin_amd64.1_13.go:27:3: //go:linkname must refer to declared function or variable
../go/pkg/mod/golang.org/x/sys@v0.0.0-20200519105757-fe76b779f299/unix/zsyscall_darwin_amd64.1_13.go:40:3: //go:linkname must refer to declared function or variable
../go/pkg/mod/golang.org/x/sys@v0.0.0-20200519105757-fe76b779f299/unix/zsyscall_darwin_amd64.go:28:3: //go:linkname must refer to declared function or variable
../go/pkg/mod/golang.org/x/sys@v0.0.0-20200519105757-fe76b779f299/unix/zsyscall_darwin_amd64.go:43:3: //go:linkname must refer to declared function or variable
../go/pkg/mod/golang.org/x/sys@v0.0.0-20200519105757-fe76b779f299/unix/zsyscall_darwin_amd64.go:59:3: //go:linkname must refer to declared function or variable
../go/pkg/mod/golang.org/x/sys@v0.0.0-20200519105757-fe76b779f299/unix/zsyscall_darwin_amd64.go:75:3: //go:linkname must refer to declared function or variable
../go/pkg/mod/golang.org/x/sys@v0.0.0-20200519105757-fe76b779f299/unix/zsyscall_darwin_amd64.go:90:3: //go:linkname must refer to declared function or variable
../go/pkg/mod/golang.org/x/sys@v0.0.0-20200519105757-fe76b779f299/unix/zsyscall_darwin_amd64.go:105:3: //go:linkname must refer to declared function or variable
../go/pkg/mod/golang.org/x/sys@v0.0.0-20200519105757-fe76b779f299/unix/zsyscall_darwin_amd64.go:121:3: //go:linkname must refer to declared function or variable
../go/pkg/mod/golang.org/x/sys@v0.0.0-20200519105757-fe76b779f299/unix/zsyscall_darwin_amd64.go:121:3: too many errors
```

puhitaku, 2022-06-20:
It looks like [this](https://github.com/google/go-jsonnet/issues/596#issue-1170475877).
And here's [the solution](https://github.com/google/go-jsonnet/issues/596#issuecomment-1072120926).

Ricotan-Linus, 2022-06-21:
after get sys, output libusb error...
but I already install libusb...

`# pkg-config --cflags  -- libusb-1.0
pkg-config: exec: "pkg-config": executable file not found in $PATH`

Ricotan-Linus, 2022-06-21:
I resolve that after install pkg-config from brew. thanks!

puhitaku, 2022-06-21:
- Please run `brew link libusb` and see if the problem persists.
 - Please make sure that you're using the correct pkg-config in your PATH. Installing the Homebrew's pkg-config may be a solution for this: `brew install pkg-config`

   ```
   $ which pkg-config
   /opt/homebrew/bin/pkg-config
   ```
 - Find `libusb-1.0.pc` in your system and add the path to `PKG_CONFIG_PATH`.
 - You're using old Mac OS X 10.13. Please keep in mind that any build error can happen in your Mac.