puhitaku/mtplvcap #64: Camera model matching doesn't work for Z cameras

oxpa, 2022-11-07:
Hello,

First of all, I'd like to confirm that Z9 does indeed work. I'm happy to provide you any data you want/need from it.
Secondly, I think I've found a small bug which doesn't affect anything.

In nikon.go there is a map where keys are camera names and values are their description.
For Z9 the product name from USB is `NIKON DSC Z 9`, for Z 7II the product name is `NIKON DSC Z 7_2`
So a slight change needed to make mtplvcap detect the model properly:
```

diff --git a/mtp/nikon.go b/mtp/nikon.go
index 2b24716..572aa66 100644
--- a/mtp/nikon.go
+++ b/mtp/nikon.go
@@ -182,12 +182,12 @@ var models = ModelMap{
                HeaderSize:     384,
                ResolutionType: ResolutionType8,
        },
-       "Z7II": {
+       "7_2": {
                Name:           "Z7II",
                HeaderSize:     384,
                ResolutionType: ResolutionType8,
        },
-       "Z9": {
+       "9": {
                Name:           "Z9",
                HeaderSize:     384,
                ResolutionType: ResolutionType8,
```

I guess, the same change will be needed for some other newer models (like z6, z7, z6ii, z5, z50, zfc). But at the same time their parameters match the default so no real benefit of doing this.
It may also worth reimplementing camera model detection and may be rely on some ID.
Sadly, I have zero experience with usb or mtp so I'm pretty useless at proving patches here.

puhitaku, 2022-11-07:
Hi, what you shared here is actually beneficial for future workarounds for per-model quirks and bugs (while I hope it doesn't exist 😌)
AFAIK I have no choice to identify the model other than this human-readable and non-standardized format string. Now I know that the D series has a `D` prefix but the Z series doesn't. (Oh dear...)

I want to ask for 2 things:

1. The diff you shared is worth being a PR. Would you create the one, or if you prefer not to do this, I'll commit as-is.

2. Many thanks for confirming that Z9 works with mtplvcap. I'd like to add a green checkmark in the README with your GitHub ID. Is it okay?

oxpa, 2022-11-07:
didn't expect a response this quick!
I've submitted #65

closed 2022-11-07 by oxpa