puhitaku/mtplvcap #66: Error on Nikon D3100

RomeoBravoGolf, 2023-01-04:
Hi, recieving the following error on Nikon D3100

```

[0000]  INFO usb: found: 04b0:0427
[0000] ERROR mtp: fatal error LIBUSB_ERROR_IO; closing connection.
[0000] FATAL main: failed to detect MTP devices: could not open 04b0:0427: mtp: no MTP extensions in ''
```
OS Ubuntu 20.04

puhitaku, 2023-01-05:
Unfortunately, [D3100 is not supported](https://github.com/puhitaku/mtplvcap#unsupported-models).

closed 2023-01-05 by puhitaku

RomeoBravoGolf, 2023-01-05:
Thanks for the quick reply

RomeoBravoGolf, 2023-01-06:
Quick follow up, did some digging and it turns out the de D3100 does have LV mode but for some reason it disables itself when entering USB mode. I have no idea why this happens only in this specific model, ive tried to circumvent this problem with everything from ffmpeg + gphoto2 to digicam Control with no success. Do you think D3000 software could somehow be installed or if there is some other way to have de original software allow LV on USB mode

puhitaku, 2023-01-06:
digiCamControl has a longer history in controlling DSLRs than mtplvcap. If digiCamControl does not support a model, we don't have any way to deal with it. There actually exists a modded firmware for Nikons. And you may disassemble it with Ghidra or something to find the flag not to stop LV when it's connected to a USB port, but I'm not sure how the modification will be. It might be a single branch instruction. Or, it might be a more complex modification of the internal state. Or, there's no feature to transmit the frame via USB.

RomeoBravoGolf, 2023-01-06:
Thanks I'll look into it