puhitaku/mtplvcap #70: Z50 hi-res report

prg318, 2023-02-11:
Hi,
I have a Nikon Z50 with the latest firmware.  I just wanted to report that I'm able to use my Z50 with `mtplvcap -max-resolution` and get 1024x576 video at about 47-50 fps.

I was surprised to achieve this - especially since the Z50 only has a micro USB connector (which I'm assuming is USB 2.x speeds).

Sorry if this is not the right place for this report, but I wanted to pass along the feedback.  Thank you for this project!

PulsarFX, 2023-02-11:
max-resolution works on my z50, too. 
Latest firmware 2.4 on the camera and linux mint 21.1 as OS with mtplvcap 1.6.1

puhitaku, 2023-02-12:
Here actually is a right place for reporting this. Thank you all!

closed 2023-02-12 by puhitaku