puhitaku/mtplvcap #71: failed to detect MTP devices: could not open 04b0:0448

dariader, 2023-07-13:
Hello, 
Thank you for the tool! 

I encounter an error after compilation 
`./mtplvcap
[0000]  INFO usb: found: 13d3:3563
[0000]  INFO usb: found: 04b0:0448
[0000]  WARN mtp: detected more than 1 device
[0000]  INFO mtp: opening the detected Nikon DSLR: 04b0:0448
[0000] FATAL main: failed to detect MTP devices: could not open 04b0:0448: failed to claim: failed to claim: LIBUSB_ERROR_BUSY`

I use Ubuntu 22.04, Nikon Z5

Any tips how to overcome this issue?

puhitaku, 2023-07-14:
Hi,
The recent Ubuntu supports MTP to transfer data from devices like Android phones. My guess is that the camera is occupied by the OS. Does the camera appear at the corner of Nautilus (file manager)?
[image]

puhitaku, 2023-09-15:
@dariader Any updates?

puhitaku, 2023-09-18:
Closing as it's stale

closed 2023-09-18 by puhitaku

HereIsZephyrus, 2025-03-09:
I encountered this problem as well and found the issue. Indeed, the problem was solved when I removed this device from the files. Thanks a lot!