puhitaku/mtplvcap #73: [macos]pkg-config: exec: "pkg-config": executable file not found in $PATH

choongmanee, 2023-09-15:
Hi,

When following macOS 2.b.4 `CGO_CFLAGS='-Wno-deprecated-declarations' go build .`, the following error was displayed:

```
➜  mtplvcap git:(master) CGO_CFLAGS='-Wno-deprecated-declarations' go build .
go build github.com/hanwen/usb:
# pkg-config --cflags  -- libusb-1.0
pkg-config: exec: "pkg-config": executable file not found in $PATH
go build github.com/google/gousb:
# pkg-config --cflags  -- libusb-1.0
pkg-config: exec: "pkg-config": executable file not found in $PATH
```

adding `pkg-config` to macOS 2.b.1 fixed the issue

puhitaku, 2023-09-15:
Thank you for the PR. I forgot to add pkg-config here.

BTW, `2.b.1` doesn't look like a valid macOS version identifier. What's the actual version? It should look like `13.5.1` (macOS Ventura, for example)

choongmanee, 2023-09-15:
Gladly! My mistake on the obscurity. The macOS version was 13.5.2. The `2.b.1` was to reference the macos install step 2b part 1

puhitaku, 2023-09-15:
Aha, understood! Thank you again!

closed 2023-09-15 by puhitaku