puhitaku/mtplvcap #76: Unable to use OBS virtual camera

tomassasovsky, 2023-11-05:
Hello there! First of all, thank you for creating such an awesome piece of software!

I'm having some trouble getting the OBS virtual camera to work on platforms like Google Meet and Zoom. I have a Nikon D5200. Here's my current config:

[image]
[image]
[image]
[image]

Any ideas on what the issue could be here? Thanks again!

puhitaku, 2023-11-06:
mtplvcap doesn't control virtual cameras which is OS-dependent.
Taking a look at your screenshots, the current video device of Google Meets is "NVID...". It is different from a usual installation of OBS virtual camera's virtual camera service.

closed 2023-11-06 by puhitaku

tomassasovsky, 2023-11-06:
Ah, very dumb mistake on my part. Thank you so much for pointing it out for me!