puhitaku/mtplvcap #80: Confirmed on Nikon D810

GlassOnTin, 2024-03-10:
Hi @puhitaku, I can confirm mtplvcap is working on my NIKON DSC D810, though I see the warning "model didn't match, falling back to the generic model":

```
:~/Code/mtplvcap$ ./mtplvcap -debug server
[0000]  INFO usb: found: 0a5c:5800
[0000]  INFO usb: found: 04b0:0436
[0000]  INFO usb: found: 413c:8187
[0000]  WARN mtp: detected more than 1 device
[0000]  INFO mtp: opening the detected Nikon DSLR: 04b0:0436
[0000]  INFO main: started
[0000] DEBUG lv: manufacturer = NIKON, product = NIKON DSC D810, serialnumber = 0000006026691
[0000] DEBUG lv: model didn't match, falling back to the generic model 
```

Let me know if there any specific tests I can help with.

GlassOnTin, 2024-03-10:
The -max-resolution option doesn't work on the Nikon D810:
```
$ ./mtplvcap -debug server -max-resolution
[0000]  INFO usb: found: 0a5c:5800
[0000]  INFO usb: found: 04b0:0436
[0000]  INFO usb: found: 413c:8187
[0000]  WARN mtp: detected more than 1 device
[0000]  INFO mtp: opening the detected Nikon DSLR: 04b0:0436
[0000]  INFO main: started
[0000] DEBUG lv: manufacturer = NIKON, product = NIKON DSC D810, serialnumber = 0000006026691
[0000] DEBUG lv: model didn't match, falling back to the generic model 
[0001]  INFO lv: getting available resolutions
[0001]  INFO lv: available resolutions (higher is larger): [1 2]
[0001]  INFO lv: automatically use the largest choice: 2
[0004] ERROR mtp: fatal error LIBUSB_ERROR_TIMEOUT; closing connection.
[0007]  WARN lv: workerLV: failed to start live view: LIBUSB_ERROR_TIMEOUT
[0007]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
[0007]  WARN lv: frameCaptor: failed to obtain an image: mtp: cannot run operation MTP_WMPPD_PlaylistObjectPref, device is not open
[0008]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
[0008]  WARN lv: frameCaptor: failed to obtain an image: mtp: cannot run operation MTP_WMPPD_PlaylistObjectPref, device is not open
[0009]  WARN lv: workerLV: failed to get live view status: mtp: cannot run operation GetDevicePropValue, device is not open
```

puhitaku, 2024-03-13:
Thanks, I'll update the table.

puhitaku, 2024-03-13:
Now it is pushed. Thank you!

closed 2024-03-13 by puhitaku