puhitaku/mtplvcap #82: Not working a lot of the time

Codepro3000, 2024-06-26:
when I go to launch the application, it does not launch. I have tried unplugging and plugging my camera back in, but it does not launch a lot of the time. The only way it works again is if i restart, even if i just powered on my PC.

puhitaku, 2024-06-27:
@Codepro3000 Please give me all the info you have: the camera's model name, the OS, how you launched mtplvcap, the log, and the debug log. The following command will launch mtplvcap with debug logs enabled: `./mtplvcap -debug server`

From past issues and experiences, I would say, this kind of instability is highly due to environmental things. The most frequent case is a bad (ingenuine) USB cable.

puhitaku, 2024-07-05:
@Codepro3000 Any update?

Codepro3000, 2024-07-11:
@puhitaku I’m using a Nikon D-90 on windows 11. Sorry for the late reply, but I am unable to access my pc right now… I can get back to you in the next week with the logs

puhitaku, 2024-09-03:
@Codepro3000 ping

puhitaku, 2025-03-09:
Stale, closing

closed 2025-03-09 by puhitaku