puhitaku/mtplvcap #86: Nikon Coolpix S4400

konstantinkorovajcev, 2025-04-12:
Lenovo Ideapad 3 15ADA05

Arch Linux x86_64

```
[0000]  INFO usb: found: 0bda:c123
[0000]  INFO usb: found: 04b0:033d
[0000]  WARN mtp: detected more than 1 device
[0000]  INFO mtp: opening the detected Nikon DSLR: 04b0:033d
[0000] ERROR mtp: fatal error LIBUSB_ERROR_PIPE; closing connection.
[0000] FATAL main: failed to detect MTP devices: could not open 04b0:033d: mtp: no MTP extensions in ''
```

When reconnected

```
[0000]  INFO usb: found: 0bda:c123
[0000]  INFO usb: found: 04b0:033d
[0000]  WARN mtp: detected more than 1 device
[0000]  INFO mtp: opening the detected Nikon DSLR: 04b0:033d
[0000]  INFO main: started
[0000] ERROR main: http server returned an error: listen tcp 127.0.0.1:42839: bind: address already in use
[0000] ERROR mtp: fatal error LIBUSB_ERROR_PIPE; closing connection.
[0000]  WARN lv: failed to obtain ISO list: LIBUSB_ERROR_PIPE
[0000]  WARN lv: failed to obtain F-values: mtp: cannot run operation GetDevicePropDesc, device is not open
[0000] ERROR main: listen tcp 127.0.0.1:42839: bind: address already in use
```

puhitaku, 2025-04-12:
AFAIK Coolpix series [does not support](https://digicamcontrol.com/cameras) the live view feature like other DSLRs do.

closed 2025-04-12 by puhitaku

konstantinkorovajcev, 2025-04-12:
I was stupid not reading this

[image]